adafruit/Adafruit-Thermal-Printer-Library #25: Adafruit Thermal Library Error

treysmith92, 2017-03-09:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **Particle PHOTON**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **Particle IDE - build.particle.io**

- Attempting to flash the adafruit_thermal code to the particle photon, which has the library included, returns errors which I've included an image of.

[image]

MartyMacGyver, 2017-04-11:
First, ignore the unmaintained version of this library in the Particle libraries.

Instead, get ready for some copy and pasting. For `examples/A_printertest`, create a project with some name, and for the main file paste the contents of `A_printertest.ino` with the following important changes:

    ..........
	// Here's the new syntax when using SoftwareSerial (e.g. Arduino Uno) ----
	// If using hardware serial instead, comment out or remove these lines:

	//xx//#include "SoftwareSerial.h"
	//xx//#define TX_PIN 6 // Arduino transmit  YELLOW WIRE  labeled RX on printer
	//xx//#define RX_PIN 5 // Arduino receive   GREEN WIRE   labeled TX on printer

	//xx//SoftwareSerial mySerial(RX_PIN, TX_PIN); // Declare SoftwareSerial obj first
	//xx//Adafruit_Thermal printer(&mySerial);     // Pass addr to printer constructor
	// Then see setup() function regarding serial & printer begin() calls.

	// Here's the syntax for hardware serial (e.g. Arduino Due) --------------
	// Un-comment the following line if using hardware serial:

	Adafruit_Thermal printer(&Serial1);      // Or Serial2, Serial3, etc.

	// -----------------------------------------------------------------------

	void setup() {

	  // This line is for compatibility with the Adafruit IotP project pack,
	  // which uses pin 7 as a spare grounding point.  You only need this if
	  // wired up the same way (w/3-pin header into pins 5/6/7):
	//xx//  pinMode(7, OUTPUT); digitalWrite(7, LOW);

	  // NOTE: SOME PRINTERS NEED 9600 BAUD instead of 19200, check test page.
	//xx//  mySerial.begin(19200);  // Initialize SoftwareSerial
	  Serial1.begin(19200); // Use this instead if using hardware serial
	  printer.begin();        // Init printer (same regardless of serial type)
    ..........

Then create two tabs (`Adafruit_Thermal.cpp` and `Adafruit_Thermal.h`) and give them the same contents as are found in the repo here. (Make sure each has the appropriate content.)

Finally, create two more tabs (`adalogo.h` and `adaqrcode.h`) and give them the same contents as in the example headers of the same names. Remove the corresponding `.cpp` tabs the IDE automatically creates for these.

The printer appears to communicate at 3.3V. Use the Photon pins labeled TX and RX (route TX on the particle to RX on the printer and vice versa). Ensure you have a solid power supply for the 5V-9V printer power (2A is recommended). Make sure the printer is on and works (hold down the black paper feed button when powering it on to generate a test page if you haven't already).

Run the sketch and it should print.

After all this, I'm considering learning how to contribute the port of this library myself...

trytofly94, 2020-12-29:
Old Topic,
But could you maybe share your app again? 
I'm desperately trying to get my thermal printer to work using Particle Photon.
Thank  you!

treysmith92, 2020-12-29:
Hey there, not sure I have the assets to that project any longer; but I did
find this bit of code saved from some point in the process, hope it helps:

 // This #include statement was automatically added by the Particle IDE.
#include <Adafruit_Thermal.h> Adafruit_Thermal printer; // String tweet;
char tweet[300]; String oldTweet; void setup(){ Serial.begin(9600); // Uses
regular serial Serial1.begin(19200); //Starts printer serial (RX (green
printer wire), TX (yellow printer wire) printer.begin(&Serial1); //uses RX
and TX Particle.subscribe("Treytest", myHandler); //subscribes to the event
printer.println(tweet); printer.feed(1); } void loop() { //empty } void
myHandler(String event, String data) { /* Particle.subscribe handlers are
void functions, which means they don't return anything. They take two
variables-- the name of your event, and any data that goes along with your
event. In this case, the event will be "buddy_unique_event_name" and the
data will be "intact" or "broken" Since the input here is a char, we can't
do data=="intact" or data=="broken" chars just don't play that way. Instead
we're going to strcmp(), which compares two chars. If they are the same,
strcmp will return 0. */ Particle.publish("myhandlerBlue Triggered",
"YES"); //Debug - let me know it was triggered int result =
data.compareTo(oldTweet); //compares to old tweet if (result == 0) {
Particle.publish("Blue_the_same", "Samesies"); } else { // if the data is
something else printReal(data); //sends this to the printing function
oldTweet = String(data); //saves this tweet in oldTweet Particle
.publish("Blue_the_same", "Not The Same"); //publish a message } } void
printReal(String thing) { //the printer function printer.wake();
printer.boldOn(); Particle.publish("thing_I_printed_blue", thing);
printer.println(thing); printer.feed(8); //8 lines are enough to clear the
opening printer.sleep(); } void print2(String event, String data) {
printer.wake(); printer.inverseOn(); printer.println(data);
printer.inverseOff(); printer.feed(8); printer.sleep(); Particle
.publish("thing_I_printed_blue", data); }

caternuson, 2021-04-08:
Closing. Seems more like a Particle IDE issue.

closed 2021-04-08 by caternuson