adafruit/Adafruit-Thermal-Printer-Library #29: Can't change codepage on firmware 2.69

zhovner, 2018-05-03:
I'm try to use another codepage by `setCodePage()`.
No matter what codepage I setup, I still get chinese symbols in upper ASCII.
But changing charset by `setCharset()` working correctly.

Here the test page I get when press button during power on: 
[image]

0julio0, 2019-01-20:
Same problem. Any sugestion?

zhovner, 2019-01-21:
@0julio0 I give up trying to fix this and use bitmap image to print custom symbols

shali379, 2019-02-02:
For some reason the Kanji mode is switched on by default. Before changing codepage, the Kanji mode must be disabled by sending the "Cancel Kanji character mode" command. The setCodePage function thus should read:
`void Adafruit_Thermal::setCodePage(uint8_t val) {`
`  if(val > 47) val = 47;`
`  writeBytes(ASCII_FS, '.'); //Cancel Kanji character mode`
`  writeBytes(ASCII_ESC, 't', val);`
`}`
I was able to set the CP1250 (West Europe) successfully this way.
Note that my printer is "VERSION 2.69 2013-11-22" and I have no idea how other printer versions will react to this command.

masteruan, 2019-02-11:
Doesn't work for me :-(

masteruan, 2019-02-12:
I've resolved my problem!! I've removed the RS232 --> TTL converter. And added some resistance under the component like in photo.
[image]

lomalkin, 2020-05-16:
> I've resolved my problem!! I've removed the RS232 --> TTL converter. And added some resistance under the component like in photo.
> [image]

Confirmed. Thank you!
I also fixed two chinese printers by removing UT3232 and adding four resistors 470R 0603.

caternuson, 2021-04-09:
Is this happening with any of the printers sold by Adafruit? This may be a region specific hardware issue?

caternuson, 2021-04-09:
Closing. Can not recreated with a [PID 2751 printer](https://www.adafruit.com/product/2751), a [Feather ESP32](https://www.adafruit.com/product/3405), and using the `B_charsets.ino` example from library.

Please provide information on the specific printer model being used.

[image]

closed 2021-04-09 by caternuson