adafruit/Adafruit-Thermal-Printer-Library #36: Error when compiling for ESP32

thearduinoguy, 2020-01-24:
When trying to upload the printer test sketch to an ESP32 I get the errors:

```
Arduino: 1.8.10 (Windows 10), Board: "Node32s, 80MHz, 921600"

In file included from C:\Users\Mike\Documents\Arduino\libraries\Adafruit_Thermal_Printer_Library\examples\A_printertest\A_printertest.ino:21:0:

C:\Users\Mike\Documents\Arduino\libraries\EspSoftwareSerial\src/SoftwareSerial.h:162:12: error: 'size_t SoftwareSerial::readBytes(uint8_t*, size_t)' marked 'override', but does not override

     size_t readBytes(uint8_t* buffer, size_t size) override;

            ^

C:\Users\Mike\Documents\Arduino\libraries\EspSoftwareSerial\src/SoftwareSerial.h:165:12: error: 'size_t SoftwareSerial::readBytes(char*, size_t)' marked 'override', but does not override

     size_t readBytes(char* buffer, size_t size) override {

            ^

Multiple libraries were found for "Adafruit_Thermal.h"
 Used: C:\Users\Mike
Multiple libraries were found for "SoftwareSerial.h"
 Used: C:\Users\Mike
exit status 1
Error compiling for board Node32s.

```

caternuson, 2020-01-27:
Do you actually need software serial? Try changing the code to hardware serial. See comments in example.

ChrisIdema, 2020-07-16:
I think you may need to remove other libraries with the same name. I got it to work on the ESP32 after installing software serial for the ESP32. Did you get it to work with hardware serial?

caternuson, 2021-04-09:
Closing. Confirmed working with hardware serial. For software serial, it may be as suggested above.

closed 2021-04-09 by caternuson