adafruit/Adafruit-Thermal-Printer-Library #38: printer.feedRows(1) prints a J in stead of a row feed, unknown firmware version

ChrisIdema, 2020-07-16:
`printer.feedRows(1); `prints a J in stead of a row feed. 
I suspected my firmware version is older, however I don't know the firmware version. Setting PRINTER_FIRMWARE in Adafruit_Thermal.h to 0 didn't change the behavior! That is strange, because it should print new-lines in case of older versions that don't support the row-feed command.
I changed the header files in both these locations:

- /home/USER/.arduino15/staging/libraries/Adafruit_Thermal_Printer_Library-1.2.0.zip
- /home/USER/Arduino/libraries/Adafruit_Thermal_Printer_Library

I added the following lines in my code as a sanity check:
```
#if PRINTER_FIRMWARE >=268
#error "PRINTER_FIRMWARE>=268"
#endif
```
It gives the error when I don't edit the header file and doesn't when I set it to 0. So that means it should be compiled using that define. I deleted the cache folder and restarted arduino ide several times. But it still prints J

I use printer #597.
This is what the sticker says:
```
Model: CSN-A2
S/N Z1912110420
```
The test page printed by the printer doesn't have a firmware version number. This is what I see:
```
Selftest
System: A2
Version: GV.16.08 20190822_R
[...]
MAX dots:80
Heat time:774
[...]
Selftest Finished
```
I have ubuntu 20.
I use esp32 DOIT devkit.
Printing text works well.
Printing bitmaps works well too, but I do see thin horizontal white lines through the image even after setting heating time to 255.

xMase, 2020-08-28:
the problem is this [command](https://github.com/adafruit/Adafruit-Thermal-Printer-Library/blob/a7dc6aeff1c44d72490dcd1c9fb3c1be1dc5068a/Adafruit_Thermal.cpp#L215) which is not interpreted by our firmware.
it would be good and right if they put the correct datasheet for our firmware.

caternuson, 2021-04-08:
Does commenting out that command fix the issue? Also not sure how #30 relates to this issue?

caternuson, 2021-04-10:
Hopefully #46 fixes this. Simply removed that offending command. If that doesn't fix it, please comment and we can reopen.

PR #46 (2021-04-10): Misc fixes.
A couple of fixes here.

* Fix #23. The busy wait loop was causing WDT reset on ESP8266. Added a `delay(0)` in the loop to allow ESP8266's implementation to yield as needed. This way don't need any pre-processor logic.
* Fix #38. Was not able to reproduce this, but based on conversation in issue thread it was related to calling an unsupported option `DC2 *` in `begin()`. Moved that configuration to a new function `setPrintDensity()` and then don't call it by default. Resulting prints look fine without it.

Here's an example WDT reset just for records:
[image]
closed 2021-04-10 by caternuson