adafruit/Adafruit-Thermal-Printer-Library #40: Carriage return 0x13 ?

HamzaHajeir, 2020-10-16:
Hi

While I'm enhancing the performance of this library by making it non-blocking, I've noticed this line : 

`if(c != 0x13) { // Strip carriage returns `

Shouldn't It be 13 Decimal, Not HEX ? 
`if(c != 13) { // Strip carriage returns`

Or someone waits Ariana 6 to happen here ? xD

Thanks

ladyada, 2020-10-16:
right! could you submit a PR? :)

HamzaHajeir, 2020-10-17:
At soonest :) 

I'll prepare my enhanced version of this library, Can you open a dev-branch to make PR to it ?

Thanks

ladyada, 2020-10-17:
you would fork it yourself
https://learn.adafruit.com/an-introduction-to-collaborating-with-version-control

HamzaHajeir, 2020-10-17:
Of-course I'd do it on my fork first, I've merged 2 forks already ;) 

But I want to do 2 separate PRs, One for CR, and one for library enhancement.

ladyada, 2020-10-17:
ok! please PR into `master`

caternuson, 2021-04-08:
Fixed with #44.

closed 2021-04-08 by caternuson