adafruit/Adafruit-Thermal-Printer-Library #52: Stuttering when printing images or test print

willmendil, 2022-03-16:
Hi,

I'm having an issue but I am not sur it's not due to the printer itself. Indeed I followed the tutorial to setup the printer with a raspberry pi and installed the zj58 drivers. However when I do the test print or print a PNG file through LP the printer stutters. That is every 5mm it stops for a second. This leads to slow prints and also to artefacts in the result.
I saw some videos on YouTube where the images are printed continuously without any issues. 

Any idea?

ladyada, 2022-03-16:
could be a bad or weak power supply

willmendil, 2022-03-16:
I'm using a bench top 30amps power supply at 9v so no problems there. When I print the test paper it printe very quickly ans smoothly

willmendil, 2022-04-05:
here's a video of the issue
https://youtu.be/1MUEXiSN3ok
As you can see, the printing stutters even though the ps has enough intensity to provide what is needed. This looks like a software issue, but I cannot find a solution. I've tried many different drivers, etc... but nothing. Is this a normal behaviour?

willmendil, 2022-04-05:
yep

willmendil, 2022-04-05:
[image]
[image]

here's the test page. the paper is bad quality so i'm using it for testing. (though i'm not excluding it is part of the issue)
i've also tried with the virtual print thing though i couldn't find online what it does, nor is it mentioned in any documentation

PaintYourDragon, 2022-04-05:
Flow control seems to be working. In the printer config dialog on the Pi, can you confirm the baud rate is set to 19,200? It looks like the printer’s receiving data in super slo-mo, hence the pauses. I mean _some_ short pause is normal even at 19,200, but this is pretty severe.

willmendil, 2022-04-05:
I've tried with an arduino mega and i get same behaviour. I also get the same stuttering when printing the test page from zj-58 driver which I tried from multiple sources. 
but yeah, it does look like it's getting the data in slow impulse. 
If i change the baud rate, i get gibberish letters printed at a correct speed

willmendil, 2022-04-05:
I followwed this https://learn.adafruit.com/mini-thermal-receipt-printer?view=all#downloads and added the DTR wire and an arduino, no it prints much faster. It's not perfect, but still better. Any idea how to incorparate this wire on a raspberry pi?

willmendil, 2022-04-06:
the AClasTool keeps serroring out no matter what i try

i tried folling this https://www.shumi-tech.online/entry/2018/07/03/Maker%E5%90%91%E3%81%91%E3%82%B5%E3%83%BC%E3%83%9E%E3%83%AB%E3%83%97%E3%83%AA%E3%83%B3%E3%82%BF%E3%80%8EDP-EH600%E3%80%8F%E3%82%92%E7%88%86%E9%80%9F%E5%8C%96%E3%81%99%E3%82%8B 
but i don't have the same printer apparently

ladyada, 2022-04-06:
@willmendil what is the exact URL of the product you purchased

willmendil, 2022-04-06:
https://www.gotronic.fr/art-imprimante-thermique-ada597-21349.htm

willmendil, 2022-04-06:
[image]

willmendil, 2022-04-12:
@ladyada any insight?

ladyada, 2022-04-12:
you could try the arduino library or not use the printer driver and just connect direct over serial. often the printers need more power, 5V or 9V @ 4A

willmendil, 2022-04-12:
tried it with an arduino mega and uno with demo codes.
What worked "better" was to use a serial to usb board and connect the DTR cable. but it was still not as good... i might have been given a bad unit... I order some other printers off aliexpress which were a bit cheaper. I'll see if they work (hopefully because this is getting expensive). I also tried uploading the firmware but i only get errors using the "AClasTool.exe tool