adafruit/Adafruit-Thermal-Printer-Library #54: printing bitmap doesn't work in example

eugeniop, 2022-08-16:
Got the Adafruit thermal printer. The example supplied works perfectly except for the 2 `printBitmap` parts.

In both cases (the adafruit logo and the QRCode) starts ok (as in the first top section of the bitmaps are printed fine), but then follows with garbled content.

I suspected a buffer overrun, so I added a `setTimes` statement to the example with 2X, and 3X the default values. There's speed difference in the printing, but no change in what is being printed.

Printer Firmware version: 2.68

- Arduino board:  tried with Feather M0 and MKR1010

- Arduino IDE version: 1.8.19

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

Used example sketch `A_printertest`.

PaintYourDragon, 2022-08-17:
Unable to reproduce. Same board, same IDE, same printer firmware.
Please check:
- Adafruit SAMD Boards package version (current is 1.7.10)
- Power supply (5–9V suggested, 2A or better)

eugeniop, 2022-08-31:
Thanks for checking!